nasa/CryptoLib #31: Crypto_TC_ApplySecurity payload CFE_PSP_MemCpy

Peco602, 2021-12-14:
There is a misalignment in payload copying from clear to encrypted frame.
Please replace:
`
CFE_PSP_MemCpy((p_new_enc_frame+index), (p_in_frame+TC_FRAME_PRIMARYHEADER_STRUCT_SIZE), tf_payload_len);
`
with:
`
CFE_PSP_MemCpy((p_new_enc_frame+index), (p_in_frame+TC_FRAME_HEADER_SIZE), tf_payload_len);
`

dccutrig, 2021-12-20:
Appears to have been in crypto.c ca. line 2912. Believe this is fixed and was caught with the updated unit tests, will check after refactor PR is merged.

IbraheemYSaleh, 2021-12-21:
I think it's line 384 in crypto_tc.c now, and still needs to be fixed.

IbraheemYSaleh, 2021-12-21:
262 has the same problem:
`memcpy(p_new_enc_frame, p_in_frame, TC_FRAME_PRIMARYHEADER_STRUCT_SIZE);`

That should also just be 5 bytes (IE `TC_FRAME_HEADER_SIZE`)

IbraheemYSaleh, 2021-12-21:
athough, I both 262 && 384 need to be variable lengths depending on if there are segmentation headers or not in the frame (which is probably why it's working with the wrong variable set to 6 bytes), so TC_FRAME_HEADER_SIZE + segment_hdr_len

Need to move segment_hdr_len declaration earlier.

dccutrig, 2021-12-22:
Handled in PR #41

closed 2021-12-22 by dccutrig