nasa/CryptoLib #44: Crypto_Key_OTAR - Fixed IV length

Peco602, 2021-12-30:
Currently, the Crypto_Key_OTAR works only with an IV with length 12 bytes.

As you can see below, the value 30 is calculating by summing the Session Key ID (2 bytes), the IV len (12 bytes) and the MAC len (16 bytes):
```c
int pdu_keys = (sdls_frame.pdu.pdu_len - 30) / (2 + KEY_SIZE);
```

Moreover, keys are extracted from the 14th byte (2 bytes for Session Key ID and 12 bytes for IV): 

```c
    gcry_error = gcry_cipher_decrypt(
        tmp_hd,
        &(sdls_frame.pdu.data[14]),                     // plaintext output
        pdu_keys * (2 + KEY_SIZE),   			 		// length of data
        NULL,                                           // in place decryption
        0                                               // in data length
    );
```

```c
for (int count = 14; x < pdu_keys; x++)
```

The IV len for OTAR should be a managed parameter so I suggest to add it as a DEFINE.

rjbrown2, 2023-10-18:
Actual data needs to be extracted from the SA, rather than the hard-coding happening above.

rjbrown2, 2024-09-04:
Currently attached to #147 
Donnie-Ice is currently working this now.

closed 2024-10-24 by Donnie-Ice