nasa/CryptoLib #72: CryptoLib ApplySecurity with Libgcrypt Fails on FIPS Mode Enabled Environment

IbraheemYSaleh, 2022-02-07:
The CryptoLib Crypto_TC_ApplySecurity function fails when using the libgcrypt cryptography interface on FIPS mode enabled environments with the following error:
`ERROR: gcry_cipher_encrypt error code 156`
`Failure: gcrypt/Invalid state`

It turns out this is because AES-GCM is not FIPS certified when manually setting the IV, like we do since the IV & Sequence Number are the same.

To get around this, we need to manually modify CryptoLib to not fail due to invalid state when FIPS mode is enabled. How exactly this should be implemented is TBD, but CryptoLib with libgcrypt should function in FIPS mode hosts.

IbraheemYSaleh, 2022-02-08:
References:
https://dev.gnupg.org/T4873
https://www.gnupg.org/documentation/manuals/gcrypt/Enabling-FIPS-mode.html

dccutrig, 2023-06-22:
Long story short: This is in testing.
https://dev.gnupg.org/T4873

Additionally, while we allow KMC to generate IVs externally if they are Null and pass them back to CryptoLib, this isn't considered FIPS compliant based on discussions in the above link. Specifically  they refer to notes within OpenSSL code, but this is the reference:

_#ifdef FIPS_MODULE
    /*
     * FIPS requires generation of AES-GCM IV's inside the FIPS module.
     * The IV can still be set externally (the security policy will state that
     * this is not FIPS compliant). There are some applications
     * where setting the IV externally is the only option available.
     */_