nasa/CryptoLib #79: IV Validation (Uniqueness, non-sequential)

dccutrig, 2022-03-03:
In a FIPS mode the IV is randomly generated, which means it may not be in a rolling / counter mode. Will need to revisit the capability to allow unique IVs without window requirements (eg not a counter mode). Also future possibility.

jlucas9, 2023-04-04:
Looks like maybe our version of libgcrypt may not have the functionality for it to manage the IV for us just yet: https://dev.gnupg.org/T4873

Need to do some more research.

dccutrig, 2023-06-22:
For Clarity: 

Pick an encryption mode with IVs (Not a Counter type mode) and ensure windowing doesn't stop pseudo-random Ivs from being accepted.

Donnie-Ice, 2024-10-15:
Update: Libgcrypt 1.11.0 is now able to generate IVs

CutrightD, 2024-10-23:
Excerpt from above:

```
For encryption of AEAD cipher modes, it should be possible to generate
an initialization vector internally within libgcrypt implementation,
in coordinated way, instead of calling @code{gcry_cipher_setiv} with
arbitrary value, so that it can ensure the security properties of AEAD
block cipher.  For this purpose, the following two functions are provided:

@deftypefun {gcry_error_t} gcry_cipher_setup_geniv (gcry_cipher_hd_t @var{h}, @
            int @var{method}, const void *@var{fixed_iv}, size_t @var{fixed_ivlen}, @
            const void *@var{dyn_iv}, size_t @var{dyn_ivlen})

Set up an initialization vector generation for AEAD cipher modes.
Generation is specified by @var{method}, fixed part of initialization
vector by @var{fixed_iv} and @var{fixed_ivlen}, and dynamic part of
initialization vector by @var{dyn_iv} and @var{dyn_ivlen}.
For @var{method}, valid values are @code{GCRY_CIPHER_GENIV_METHOD_CONCAT}
and @code{GCRY_CIPHER_GENIV_METHOD_XOR}.
@end deftypefun

@deftypefun {gcry_error_t} gcry_cipher_geniv (gcry_cipher_hd_t @var{h}, @
            void *@var{iv}, size_t @var{ivlen})

Generate the initialization vector into the output buffer @var{iv}
with length @var{ivlen}.  The initialization vector will be used by
following @code{gcry_cipher_encrypt} call.
@end deftypefun
```

Reading through code, the only implementation supported appears to be 'GCRY_CIPHER_GENIV_METHOD_CONCAT' , the relevant excerpt is:

```
  memcpy (iv, hd->aead.fixed, hd->aead.fixed_iv_len);
  memcpy ((byte *)iv+hd->aead.fixed_iv_len,
          hd->aead.dynamic, hd->aead.dynamic_iv_len);
  rc = hd->mode_ops.setiv (hd, iv, iv_len);

  for (i = hd->aead.dynamic_iv_len; i > 0; i--)
    if (++hd->aead.dynamic[i - 1] != 0)
      break;
```

Donnie-Ice, 2024-10-24:
Related to nasa/cryptolib#72

Donnie-Ice, 2025-03-17:
We should be able to test on [Alma Linux](https://almalinux.org/blog/2024-10-07-fips-validation-for-almalinux-update/#:~:text=Anyone%20who%20uses%20AlmaLinux%209.2%20can%20achieve%20FIPS,dnf%20-y%20install%20openssl-3.0.7-20.el9_2.tuxcare.1%20kernel-5.14.0-284.11.1.el9_2.tuxcare.5%20fips-mode-setup%20--enable%20reboot)

Donnie-Ice, 2025-04-02:
https://csrc.nist.gov/projects/cryptographic-module-validation-program/validated-modules/search?SearchMode=Basic&ModuleName=gcrypt&CertificateStatus=Active&ValidationYear=0

Donnie-Ice, 2025-06-23:
Seems like capstone is using fips mode with a custom wolfssl wrapper successfully