nasa/CryptoLib #89: Add Github Unit Tests for Cryptography - KMC Crypto Interface

IbraheemYSaleh, 2022-03-16:


IbraheemYSaleh, 2022-03-16:
Nearly every time the Cryptography interface changes or has features added to it, the KMC Crypto interface implementation must change as well...

However, because there are no KMC Crypto Interface unit tests in the CryptoLib build flows, the interface always breaks and requires manual fixing and testing by the JPL team.

We need to add and setup a KMC Crypto Service instance to our unit test SQL scripts and run unit tests... Note, for this unit test flow, we need to have a simple non-mTLS, non-CAM tomcat service that hosts the KMC Crypto Service WAR setup locally. This should not be integration testing using CAM/mTLS/JPL hosted KMC!

rjbrown2, 2022-10-11:
I'm still working to get the KMC container added in.  It is complete locally.  However, we do not have the privileges to access the project settings tab that would allow me to add secret variables and credentials to keep the not yet open source KMC container secure.  I've sent out some e-mails and questions as to how we can accomplish this for our project.

rjbrown2, 2023-12-13:
Closing this in favor of #206.  This issue number will be referenced there, as these are duplicate issues.

closed 2023-12-13 by rjbrown2