nasa/CryptoLib #133: CryptoLib Should Provide a Function to Properly Free TC_t Types

IbraheemYSaleh, 2022-10-20:
Downstream applications using CryptoLib are expected to allocate memory for a TC_t (needed for ProcessSecurity calls):

`extern int32_t Crypto_TC_ProcessSecurity(uint8_t* ingest, int *len_ingest, TC_t* tc_sdls_processed_frame);`

However, a TC_t is then composed of multiple different CryptoLib structs, some of which (EG `TC_FrameSecurityHeader_t`). Some of these subtypes then malloc individual memory blocks as Crypto_TC_ProcessSecurity works with them, EG:

`tc_sdls_processed_frame->tc_sec_header.iv = calloc(1,sa_ptr->iv_len);`

Simply freeing the TC_t does not properly free the subsequently allocated memory blocks in the tc_sec_header... Hence, CryptoLib needs to provide a function to properly free TC_t types, EG:
`Crypto_Free_TCt(TC_t* processed_security_frame)`

IbraheemYSaleh, 2022-10-20:
Which raises this problem: https://stackoverflow.com/questions/25904993/how-to-check-if-a-struct-is-null-in-c-or-c

The substructs (tc_header, tc_sec_header, tc_sec_trailer) have no way of knowing whether or not they're empty.

jlucas9, 2023-06-15:
Need to review and see if this was also resolved as part of https://github.com/nasa/CryptoLib/issues/164