nasa/CryptoLib #134: How the Space Data Link Security (SDLS) protocol is implemented?

Honourable-A, 2022-10-29:
Hi,
I am trying to understand the CryptoLib and cFS. But I am not still clear how the SDLS protocol is implemented in this project. If you can tell me the components or modules or files where this protocol is implemented and how it is enforced that will give me a better idea. Also, if you can give me an example how this CryptoLib can be used that will also help me in my understanding.

matz96, 2022-11-17:
Hi
I am also trying to understand the CryptoLib and its implementation in the cFS. Any recomendations on how and where the Cryptolib can and should be implemented in the cFS would be helpful. If the CryptoLib has not yet been implemented in the cFS it would be helpufl to know that to, so that I can come up with my own solution.
 @rjbrown2 @IbraheemYSaleh @dccutrig 

Kind Regards
matz96

iysaleh, 2022-11-18:
As I understand it, CryptoLib was initially developed with cFS integration to help verify the CCSDS SDLS specification -- and it succeeded for that purpose, however, I don't think it was an operationally complete implementation at the time -- just an experimental proof of concept.

This past year, we've primarily developed CryptoLib to be a standalone SDLS-implementing library with a focus on telecommanding (TC Frames) so that it can be integrated into various ground data systems -- this has also been a successful effort and the implementation has been proven operationally with flight systems. 

The library has changed significantly since its initial implementation and the cFS integration hasn't really been maintained (that hasn't been our goal). It probably wouldn't be a ton of work to get cFS integration functional with the latest iteration of CryptoLib, but I don't think it'll work unless some development effort is undertaken.

The NASA IV&V team would have more information about cFS integration and any plans related to that. (@jlucas9  @szemerick  @rjbrown2  )

Honourable-A, 2022-11-18:
Hi,
Thanks for your response and giving some clarification. You mentioned that the library is now developed to be integrated into various ground data systems. Can you please elaborate on this part like which ground data system can use this library now? Is this system open source? Can we test this library if the system is not available to public? Thanks

jlucas9, 2023-03-23:
Closing for now but providing reference to original documentation about CryptoLib - https://researchrepository.wvu.edu/cgi/viewcontent.cgi?article=8150&context=etd

closed 2023-03-23 by jlucas9