nasa/CryptoLib #136: Branch Cleanup

jlucas9, 2022-11-14:


jlucas9, 2023-03-23:
@rjbrown2 can we remove the following branches:
* 88-add-github
* analysis_test
* MMCSCTTL-70
* performance_debugging

jlucas9, 2023-03-23:
@dccutrig can we remove the following branches:
* 100-hmac
* hmac_addition
* 81-add-encryption-only
* arsn_updates
* sca_bug_fixes

rjbrown2, 2023-04-13:
> @rjbrown2 can we remove the following branches:
> 
> * 88-add-github
> * analysis_test
> * MMCSCTTL-70
> * performance_debugging

88 is stale, but we might use it in the future, it was never merged into main, as it was when I was playing with the external KMC containers.

Analysis test can go.  That was something I was just playing with.

The MMCS-70 Branch was Ibraheem, it does not look to have been merged in, and is a cmake fix

closed 2023-04-28 by jlucas9