nasa/CryptoLib #138: Remove Unnecessary Libgcrypt Version Check

IbraheemYSaleh, 2023-01-06:
While using the libgcrypt interface, you get a version check error message if not using V1.9.4.

This version check & error message are unnecessary and should be removed:
https://github.com/nasa/CryptoLib/blob/dev/src/src_cryptography/src_libgcrypt/cryptography_interface_libgcrypt.template.c#L549

dccutrig, 2023-06-16:
Not sure that this needs done - although it should only check if using libgcrypt. I believe there was a specific reason we landed on that as a minimum version, although I don't recall what it was or that we communicated it to Ibraheem. Perhaps @rjbrown2 recalls

jlucas9, 2023-06-22:
https://github.com/nasa/CryptoLib/blob/dev/src/crypto/libgcrypt/cryptography_interface_libgcrypt.template.c#LL108C1-L109C1
^ Updated link

jlucas9, 2023-07-10:
Leaving in for now to ensure built version of Gcrypt matches running version of libgcrypt. Can reconsider again if another issue arises.

closed 2023-07-10 by jlucas9