nasa/CryptoLib #143: Completion of error handling

elfring, 2023-02-23:
Would you like to add more error handling for return values from functions like the following?
* [fseek](https://pubs.opengroup.org/onlinepubs/9699919799/functions/fseek.html "Reposition a file-position indicator") ⇒ [c_read_file](https://github.com/nasa/CryptoLib/blob/1c06f1601020886d7a5da0f98ab601d89506740d/util/core/shared_util.c#L39)
* [malloc](https://pubs.opengroup.org/onlinepubs/9699919799/functions/malloc.html "Memory allocation") ⇒ [python_cmac](https://github.com/nasa/CryptoLib/blob/a259545996c1f880e2d992614ed8f97b10f355e7/util/src_util/et_dt_validation.c#L100)

dccutrig, 2023-06-22:
Closing due to age.

closed 2023-06-22 by dccutrig

elfring, 2023-06-22:
:thought_balloon: I suggest to avoid [ignorance of return values](https://cwe.mitre.org/data/definitions/252.html "Checking more return values") a bit more.
Would you like to detect every error situation as early as possible?