nasa/CryptoLib #189: Change/check free(pointer) pattern

dccutrig, 2023-07-20:
e.g. instead of free(aad) look at if(!aad) free(aad);

Identified a possible issue in TC_Apply if using an AEAD algorithm for encryption only. Need to assess further, deals with AAD not being generated for encryption only, but being freed in error cases.

jlucas9, 2023-07-21:
I think we may just want to `#define` the maximum length for all fields and have them declared instead of mallocing / freeing anywhere if possible.

dccutrig, 2023-07-23:
We could, but there was a lot of byte math that would need re-worked to support this. Particularly (If i recall) in the interactio nand ordering of byte fields sent back and forth to KMC.

dccutrig, 2023-07-27:
Robert and I walked through the GSFC setup and determined that the logic for what they were trying to do exists, seems it was just a config issue.

jlucas9, 2023-07-27:
Fix merged into `dev` already and applied to a new `kmc` branch for continued NASA JPL testing.

closed 2023-07-27 by jlucas9

spicydll, 2023-10-24:
This is an anti-pattern and introduces memory leaks. `free()` does nothing if it is passed a null pointer. What this pattern does is only free pointers if they are null, and therefore already failed to allocate. Therefore, any successfully allocated heap pointer will never be free'd when using this pattern. I highly recommend you revert this.