nasa/CryptoLib #190: Consider single return per statement

dccutrig, 2023-07-20:
Originally layout made sense. Making issue for internal discussion of having just one return per function, perhaps utilizing gotoreturn labels. ISsue to be discussed / closed / modified as appropriate.

Donnie-Ice, 2024-06-11:
Put a halt on this due to rapidly increasing cyclomatic complexity. If you replace the early returns with a status message that gets returned at the end, you must add a ton of 'if' statements checking if the function should be returned.

rjbrown2, 2024-09-04:
We may want to revisit now that we've reduced complexity once (Mainly because we still have to revisit on the KMC side once open source containers are released).
If we can provide single returns, and then break functions up into smaller pieces as they make sense, we can reduce complexity again, and probably improve test coverage.

Donnie-Ice, 2025-07-14:
Working this into https://github.com/nasa/CryptoLib/issues/453

closed 2025-09-25 by Donnie-Ice