nasa/CryptoLib #196: Reducing cyclomatic complexity

jlucas9, 2023-08-31:
Need to generate cyclomatic complexity and reduce it to accommodate NASA Class B FSW standards.

rjbrown2, 2023-10-06:
We've requested this, and static analysis through the IVV SWAT team.  Have not heard back after the initial request.  Provided them with the URL to the repository.  Just awaiting a response.  Will check back in next week if we haven't heard anything by then.

rjbrown2, 2023-10-12:
Touched base with SWAT again.  Still awaiting setup on this.

rjbrown6, 2023-12-13:
Complexity format has been updated on IVV side.  Readable results now.  Have started a spreadsheet with these initial results.  Many things are not all that complex, and should remain as they are.  However, our Apply/Process functions could use some discussion.

Have put in a request with IVV to rerun our job with the latest dev branch, rather than just main.  This should give us some more results.

jlucas9, 2024-03-20:
These changes are getting very large. May be worth the time to slowly chunk these up into small pieces for easier merge review. Ideally different issues and merges for each section you're refactoring.

rjbrown2, 2024-03-20:
Will create new issues for the remaining refactors first.  If we want to separate the currently refactored portions, I can do that as well.

rjbrown2, 2024-03-20:
For additional reference, the following functions have been refactored within this issue:
crytpo_tc:
- Crypto_TC_Apply_Security_Cam
- Crypto_TC_Process_Security_Cam

sadb_routine_inmemory_template.c:
- sadb_get_operational_sa_from_gvcid

crypto.c:
- Crypto_Check_Anti_Replay

cryptography_interface_lib_gcrypt_template.c:
- cryptography_aead_encrypt
- cryptography_validate_authentication
- cryptography_aead_decrypto.c

base64url.c:
- base64urlDecode

standalone.c:
- crypto_standalone_tm_process

rjbrown2, 2024-03-20:
Currently, there is also some crypto_tm.c refactoring (tm_apply) - but it is not yet finished.  I'm going to leave at least the one partially finished in this branch.  I'll probably move the process functionality to its own branch.

rjbrown2, 2024-04-04:
Closing this issue as it is complete.  However, for tracking purposes:
#230 #231 #228 #227 #226

closed 2024-04-04 by rjbrown2