nasa/CryptoLib #201: NULL_IV in KMC Crypto

dccutrig, 2023-10-05:


dccutrig, 2023-10-05:
Tracking for an issue that has to do with CryptoLib. CrpytoLib or KMC seem to be replacing a NULL IV with a 0, which is inappropriate behavior.

Working on further dxing this issue.

Robert has access to build environment for direct testing, able to build locally.

Reaching out to Ken & Kam for follow-up on recreating their test case.

rjbrown2, 2023-10-06:
Have the ability to rebuild, and relink the shared objected (libCrypto.so), however, having issues within the VM and seeing the service pick up the changes.  I've pinged the JPL team for assistance on Monday, as they are off today.  IVV team all looked at the issue and confirmed things do not appear to be working as intended, and we're going to need some assistance.

rjbrown2, 2023-10-06:
Met with Kam and Ken.  Modified several libCrypto.so files and were unable to get the build to work as anticipated.  Ken reset everything back to their original non-debug build.  Everything handled the NULL-IV (or seemed to).  We're still going to tag up on Monday to discuss, and figure out why.  I'm going to point to a debug message causing the segfault at this point, if their normal build is working.

dccutrig, 2023-10-11:
W/ Robert: Looks like Debug print statements don't have Null logic checks. After this is fixed, either leave open, or if need to close make a similar issue for AOS so I can circle back and track it.

rjbrown2, 2023-10-12:
We need to move things back into DEV. Going to hold off fixing the NULL Print issue until the branches are stable again - to avoid duplication of work.

jlucas9, 2023-10-26:
This is good to go, @dccutrig will also resolve in his AOS branch

closed 2023-10-26 by jlucas9