nasa/CryptoLib #224: PDU length bits vs bytes

martinerikwerner-aac, 2024-03-03:
CCSDS 355.1-B-1 indicates that the length field of EP PDUs should be the length of the PDU data field in **bits**:

> **5.3.2.3.1** The Extended Procedures Data Field Length shall signal the length of the Extended Procedures PDU Data Field in bits.

This is also indicated by the declaration in:

https://github.com/nasa/CryptoLib/blob/ef64d8d1346ead848415939a21beb8dfa9fe29d3/include/crypto_structs.h#L107

However, the use of this field seems to treat is as the number of octets, for example:

https://github.com/nasa/CryptoLib/blob/ef64d8d1346ead848415939a21beb8dfa9fe29d3/src/core/crypto_mc.c#L169-L171

https://github.com/nasa/CryptoLib/blob/ef64d8d1346ead848415939a21beb8dfa9fe29d3/src/core/crypto.c#L270-L271

rjbrown2, 2024-03-07:
Hi [martinerikwerner-aac](https://github.com/martinerikwerner-aac).  Thanks for bringing this to our attention.  We have not yet worked on Extended procedures at this time - so the tests are currently broken.  This is on our short-term roadmap, and we will leave this issue open for tracking and completion.  Thanks again!

Donnie-Ice, 2024-10-01:
Linked to [nasa/cryptolib#146] [nasa/cryptolib#147] [nasa/cryptolib#148]

closed 2024-10-17 by Donnie-Ice