nasa/CryptoLib #247: CryptoLib and CFS help

Simon9050, 2024-05-31:
Hi,

I’m interested in testing the CryptoLib with CFS during a NOS3 simulation. However, I’ve been unable to find all the necessary information to perform a simple test. I am not at all an expert in Cryptography, but I would like to be able to see the effect of encryption on TC/TM frames. 

Could you please provide guidance on how to activate and observe the behavior when the CryptoLib is enabled in NOS3 ? 
(For now I am not using the latest version of NOS3, but I will try to switch to the new one in the coming weeks).

Thank you for your help and for your work.

Best regards,

Simon

jlucas9, 2024-06-03:
Hi @Simon9050,
On the latest main branch or NOS3 release 1.7.0 CryptoLib should be running both inside of cFS as well as in "standalone" with the ground software. By default in this version of NOS3, the uplink to the spacecraft will be in TC frames while the downlink will simply be Space Packet Protocol (sometimes just referred to as CCSDS) packets. Let me try to trace out and provide a short walkthrough:
* Ground software sends a command through the radio interface, preparing the Space Packet for transmission
  * Note that CryptoLib is only in use through the radio, the debug mimics direct access as if you're plugged into the satellite
* GSW forwards the space packet to "standalone" CryptoLib
* "Standalone" CryptoLib receives the packet and encapsulates it in a Telecommand or TC frame using the active spacecraft configuration, by default it's virtual channel ID 1 or clear text mode
* "Standalone" CryptoLib sends the TC frame to the spacecraft
  * Note traditionally this would be through an RF interface using a dish, but we're just using UDP in the virtual environment
* The command ingest (CI) application in cFS receives the TC frame, interprets it, and calls the cFS side CryptoLib TC Process Security function to determine if any decrypting needs done and remove all the extra security headers
* CI then places the space packet protocol message on the software bus to be routed to it's final destination

In order to enable seeing the data easily, with everything up and running I'd recommend the following:
* `sc_1 -CryptoLib` terminal
  * help
  * tc
  * vcid 1
    * Clear mode (default)
* Command Sender
  * cmd("CFS_RADIO CFE_ES_NOOP")
* `sc_1 -CryptoLib` terminal
  * vcid 4
    * Encrypted mode
* Command Sender
  * cmd("CFS_RADIO CFE_ES_NOOP")

You will first see the NOOP command received in the "standalone" CryptoLib and then passed as plain text, but received by the flight software followed by the same command being encrypted and still interpreted by the flight software successfully. 

[image]

Please feel free to ask any follow up questions here and the team and I will try to get back to as soon as possible!

Simon9050, 2024-06-05:
Thank you very much for your help, 

I was able to see the data following your instructions with NOS3 1.7.0  :) 

As I have modified a NOS3 component (EPS) + 42 in a previous NOS3 release (1.6.2), I was wondering if there is a similar simple set up to see the data or if this is only possible in 1.7.0.

Thanks, 

Simon

jlucas9, 2024-06-05:
Hi @Simon9050,
I think the release of NOS3 1.6.2 had command encryption enabled as well. You'd likely need to be sure to checkout the correct versions of CI, TO, IO_Lib, and CryptoLib itself to ensure they all sync up and work together with that specific version of cFS. How you launch and control the standalone CryptoLib may have minor changes as well.

jlucas9, 2024-06-28:
Closing due to age, please reopen or file a new issue for you have more questions!

closed 2024-06-28 by jlucas9