nasa/CryptoLib #375: Examine TLV bounding

dccutrig, 2024-12-17:
Examine TLV bounding in Crypto_Process_Extended_Procedure_Pdu.

Need to ensure that TLV values in a packet are:
1) Less than or equal to Max TLV Length (done, and fixed otherwise segfault - UTs exist) , and
2) That TLV values make sense in the context of the length of both frames and space packets received

2 is a little more complicated and I can revisit

dccutrig, 2024-12-18:
Behold, an example of what mismatched lengths and corrupted values can currently do - and also the point of this issue:

[image]

Donnie-Ice, 2025-02-11:
Need to look deeper into sa_start

closed 2025-04-02 by Donnie-Ice