nasa/CryptoLib #405: Builds Failing on Mac's

Donnie-Ice, 2025-02-06:
Robert reported that CryptoLib does not build on Mac, specifically with the custom changes to the Nos3 CryptoLib version.

jlucas9, 2025-02-07:
Not actually CryptoLib but the required steps in the Dockerfile, specifically the gpgerror and libgcrypt steps.

jlucas9, 2025-02-07:
```
35.60 mv -f .deps/t-secmem.Tpo .deps/t-secmem.Po
35.60 /bin/bash ../libtool  --tag=CC   --mode=link gcc -I/usr/local/include -g -O2 -fvisibility=hidden -fno-delete-null-pointer-checks -Wall -no-install  -o t-secmem t-secmem.o ../src/libgcrypt.la ../compat/libcompat.la  
35.63 libtool: link: gcc -I/usr/local/include -g -O2 -fvisibility=hidden -fno-delete-null-pointer-checks -Wall -o t-secmem t-secmem.o  ../src/.libs/libgcrypt.so ../compat/.libs/libcompat.a -Wl,-rpath -Wl,/tmp/libgcrypt-1.11.0/src/.libs
35.64 /usr/bin/ld: ../src/.libs/libgcrypt.so: undefined reference to `gpgrt_logv_domain@GPG_ERROR_1.0'
35.64 /usr/bin/ld: ../src/.libs/libgcrypt.so: undefined reference to `gpgrt_add_post_log_func@GPG_ERROR_1.0'
35.64 collect2: error: ld returned 1 exit status
35.64 make[1]: *** [Makefile:859: t-secmem] Error 1
35.64 make[1]: Leaving directory '/tmp/libgcrypt-1.11.0/tests'
35.64 make: *** [Makefile:500: install-recursive] Error 1
------
WARNING: current commit information was not captured by the build: failed to read current commit information with git rev-parse --is-inside-work-tree
Dockerfile:75
--------------------
  74 |     ARG GCRYPT_VERSION=1.11.0
  75 | >>> RUN curl \ 
  76 | >>>     -LS https://www.gnupg.org/ftp/gcrypt/libgpg-error/libgpg-error-${GPG_ERROR_VERSION}.tar.bz2 \
  77 | >>>     -o /tmp/libgpg-error-${GPG_ERROR_VERSION}.tar.bz2 \
  78 | >>>     && tar -xjf /tmp/libgpg-error-${GPG_ERROR_VERSION}.tar.bz2 -C /tmp/ \
  79 | >>>     && cd /tmp/libgpg-error-${GPG_ERROR_VERSION} \
  80 | >>>     && ./configure \
  81 | >>>     && make install \
  82 | >>>     && curl \ 
  83 | >>>         -LS https://www.gnupg.org/ftp/gcrypt/libgcrypt/libgcrypt-${GCRYPT_VERSION}.tar.bz2 \
  84 | >>>         -o /tmp/libgcrypt-${GCRYPT_VERSION}.tar.bz2 \
  85 | >>>     && tar -xjf /tmp/libgcrypt-${GCRYPT_VERSION}.tar.bz2 -C /tmp/ \
  86 | >>>     && cd /tmp/libgcrypt-${GCRYPT_VERSION} \
  87 | >>>     && ./configure \
  88 | >>>     && make install \
  89 | >>>     && ldconfig \
  90 | >>>     && rm -rf /tmp/*
  91 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c curl     -LS https://www.gnupg.org/ftp/gcrypt/libgpg-error/libgpg-error-${GPG_ERROR_VERSION}.tar.bz2     -o /tmp/libgpg-error-${GPG_ERROR_VERSION}.tar.bz2     && tar -xjf /tmp/libgpg-error-${GPG_ERROR_VERSION}.tar.bz2 -C /tmp/     && cd /tmp/libgpg-error-${GPG_ERROR_VERSION}     && ./configure     && make install     && curl         -LS https://www.gnupg.org/ftp/gcrypt/libgcrypt/libgcrypt-${GCRYPT_VERSION}.tar.bz2         -o /tmp/libgcrypt-${GCRYPT_VERSION}.tar.bz2     && tar -xjf /tmp/libgcrypt-${GCRYPT_VERSION}.tar.bz2 -C /tmp/     && cd /tmp/libgcrypt-${GCRYPT_VERSION}     && ./configure     && make install     && ldconfig     && rm -rf /tmp/*" did not complete successfully: exit code: 2
```
From nos3 deployment branch nos3#352

jlucas9, 2025-02-11:
Pushed up a `dev` image, but still having an error on internal:
```
[ 35%] Linking C executable standalone
/usr/bin/ld: /usr/local/lib/libgcrypt.so: undefined reference to `gpgrt_logv_domain@GPG_ERROR_1.0'
/usr/bin/ld: /usr/local/lib/libgcrypt.so: undefined reference to `gpgrt_add_post_log_func@GPG_ERROR_1.0'
collect2: error: ld returned 1 exit status
make[2]: *** [support/CMakeFiles/standalone.dir/build.make:98: support/standalone] Error 1
make[1]: *** [CMakeFiles/Makefile2:947: support/CMakeFiles/standalone.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
```